FriendsOfCADability/CADability #24: DimensionStyleSelectionPropertyException occurs when creating a new Dimension

dsn27, 2021-02-11:
```
CADability.UserInterface.DimensionStyleSelectionPropertyException
  HResult=0x80131600
  Message=no appropriate DimensionStyle available
  Source=CADability
  StackTrace:
   at CADability.UserInterface.DimensionStyleSelectionProperty..ctor(String resourceId, DimensionStyleList list, IDimensionStyle dimensionStyle, EDimType dimType, Boolean includeUndefined) in C:\Development\Repos\SOFAgh\CADability\CADability\DimensionStyleSelectionProperty.cs:line 45
   at CADability.GeoObject.Dimension.GetAttributeProperties(IFrame Frame) in C:\Development\Repos\SOFAgh\CADability\CADability\Dimension.cs:line 2205
   at CADability.Actions.ConstructAction.OnSetAction() in C:\Development\Repos\SOFAgh\CADability\CADability\ConstructAction.cs:line 6424
   at CADability.Actions.ConstrDimensionPoints.OnSetAction() in C:\Development\Repos\SOFAgh\CADability\CADability\ConstrDimensionPoints.cs:line 238
   at CADability.ActionStack.SetAction(Action Action) in C:\Development\Repos\SOFAgh\CADability\CADability\ActionStack.cs:line 118
   at CADability.FrameImpl.SetAction(Action Action) in C:\Development\Repos\SOFAgh\CADability\CADability\Frame.cs:line 781
   at CADability.FrameImpl.OnCommand(String MenuId) in C:\Development\Repos\SOFAgh\CADability\CADability\Frame.cs:line 1318
   at CADability.Forms.CadFrame.OnCommand(String MenuId) in C:\Development\Repos\SOFAgh\CADability\CADability.Forms\CadFrame.cs:line 71
   at CADability.Forms.ToolBars.TagInfo.CADability.UserInterface.ICommandHandler.OnCommand(String MenuId) in C:\Development\Repos\SOFAgh\CADability\CADability.Forms\ToolBars.cs:line 43
   at CADability.Forms.MenuItemWithHandler.OnClick(EventArgs e) in C:\Development\Repos\SOFAgh\CADability\CADability.Forms\MenuManager.cs:line 173
   at System.Windows.Forms.MenuItem.MenuItemData.Execute()
   at System.Windows.Forms.Command.Invoke()
   at System.Windows.Forms.Command.DispatchID(Int32 id)
   at System.Windows.Forms.Control.WmCommand(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.ToolStrip.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.DebuggableCallback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)
   at System.Windows.Forms.UnsafeNativeMethods.DispatchMessageW(MSG& msg)
   at System.Windows.Forms.Application.ComponentManager.System.Windows.Forms.UnsafeNativeMethods.IMsoComponentManager.FPushMessageLoop(IntPtr dwComponentID, Int32 reason, Int32 pvLoopData)
   at System.Windows.Forms.Application.ThreadContext.RunMessageLoopInner(Int32 reason, ApplicationContext context)
   at System.Windows.Forms.Application.ThreadContext.RunMessageLoop(Int32 reason, ApplicationContext context)
   at System.Windows.Forms.Application.Run(Form mainForm)
   at CADability.App.Program.Main(String[] args) in C:\Development\Repos\SOFAgh\CADability\CADability.App\Program.cs:line 19

```

SOFAgh, 2021-02-18:
fixed (was a bug in JSON deserialisation)

dsn27, 2021-05-14:
The issues seems to have come back in the latest version.
Same error message and stack trace.

SOFAgh, 2021-05-31:
Sorry, fixed again

closed 2021-05-31 by SOFAgh