FriendsOfCADability/CADability #49: Fix all compile warnings

stefan-tb, 2021-10-09:
Ideally, one should fix all the compile warnings. Especially all "obsolete" warnings in the CADabilty code base itself should be solved by shaping CADability legacy code into the desired patterns as a migration reference for other contributors.

ChrisClems, 2024-12-07:
Would it be helpful if I started tackling some of the easy ones? I looked through a handful of them and it looks like most of them are unused fields, events, and variables that date back to the first commit.

dsn27, 2024-12-08:
That would be helpful, but please don’t spend too much time on it. 
I believe there are larger priorities to focus on.

ChrisClems, 2024-12-08:
It's really just an excuse to dig into the codebase and learn more about it. I have a couple big projects in the works that I'll be using this for so I'd like to contribute but I don't have a ton of experience collaborating with other programmers so I'd like to ease into it. Cleaning up unused variables seems like an easy win and I can learn more about the structure and do some translations in the process.

dsn27, 2024-12-09:
Your help is really appreciated. Thanks a lot for spending time on this project.

stefan-tb, 2024-12-09:
As the one who opened this: removing private variables is fine. Doing more, especially working on the `[obsolete]` parts gets tricky quickly. The code base is half-way trough a refactoring which was never finished. That said, any decrease in warnings is highly appreciated.

We ourselves might be able to contribute a substantial rewrite of the UI and rendering parts down the line, splitting Math and UI entirely. I really want to see this fly.

dsn27, 2024-12-12:
- [ ] CS DialogDebuggerVisualizer.DialogDebuggerVisualizer()' is obsolete: 'Visualizers that target Visual Studio 17.6 or newer should use the other DialogDebuggerVisualizer overload and use a non-legacy formatting policy for improved security. #192
- [ ] Remove unused fields CS0169
- [ ] Remove unused variables CS0219
- [ ] Fix CS0219 The variable is assigned but its value is never used
- [ x] Fix CS0162 Unreachable code detected

Fixing these issues shouldn't have a big impact on the program itself and are easy to fix.

- [ ] CS0618 Remove obsolete properties.
This is going to be hugh. Every single Property/Function has to be checked.

- Is the replacement working properly?
- Are there parts of the program that still rely on this Property/Function?

We should probably create an issue and PR for each single Property/Function that will be replaced.

12th Dec - 970 Warnings

dsn27, 2025-02-28:
If this PR goes through we are down to 87 warnings.

https://github.com/FriendsOfCADability/CADability/pull/245

stefan-tb, 2025-02-28:
Highly appreciated!

dsn27, 2025-03-19:
All warnings have been resolved, except for those related to DebuggerVisualizer, which will no longer be addressed.

In #192, I recommended adopting Microsoft's new approach for DebuggerVisualizers.

closed 2025-03-19 by dsn27