FriendsOfCADability/CADability #64: Ruled surface with a point

happydpc, 2022-01-07:
Here's the case that to make a eccentric cone which can't make with a ruled surface.

[image]

SOFAgh, 2022-01-07:
You need two paths or curves to make a ruled solid. A point as a part is not implemented but can easily be implemented if needed. I tried with two eccentric circles, which works
[image]
.

SOFAgh, 2022-01-07:
See ruledSurface.cs:
`    /// <summary>
    /// Surface definition of a surface defined by two curves. Both curves use the standard parameter interval from 0.0 to 1.0
    /// The u-direction is provided by a combination of the two curves. The v parameter is defined by a line starting on the
    /// first curve and ending on the second curve. It is the surface described by a wire or rubber band synchronously moving along
    /// the two curves. the default parameter space is 0.0 to 1.0 on u and v.
    /// </summary>
    [Serializable()]
    public class RuledSurface : ISurfaceImpl, ISerializable, IExportStep
    {
 `

happydpc, 2022-01-07:
Yes, that works. But with cone, it fails, I am trying to test the code, especially the triangulation algorithm, now this is not a path any more.

SOFAgh, 2022-01-07:
Let me know if you need more information

happydpc, 2022-01-07:
Ok, I will test some testcases from OCC. This repo is wonderful, hope that we can make it better. And here's the question:

1. What's the "BoxedSurfaceEx" using for in Surface.PositionOf  ?

2. What's the GetNaturalBounds for? 

3. Why there's a triangulation step in ProjectedCurve? 

I'm afraid that there will be some may-be easy questions.

[image]
[image]

SOFAgh, 2022-01-08:
Hi, here is a short answer, since I am on vacation for a week:

The BoxedSurfacesEx: the surface is subdivided into patches. Each patch is wrapped (enclosed) by a parallelepiped. So if we need for example the u/v coordinates of a 3d point on a surface, we first find the specific parallelepiped which contains the point, then we are already close to the u/v coordinate and can do the remaining steps by a GaussNewton approximation. Of course some surfaces override the PositionOf, because they can do it faster. (Maybe there is a better way to calculate PositionOf for a ruled surface, but since all surfaces implement the BoxedSurfaceEx, and BoxedSurfaceEx implements PositionOf, and I am lazy, I didn't implement it)

Some surfaces have natural bounds (e.g. SphericalSurface, NURBS surface, but not the CylindricalSurface, which is infinite in one direction) and the u/v coordinate must be inside these bounds.

Similar the projected curve (wich is 2d) is enclosed by triangles to enable fast intersection or other geometric operations.

I appreciate very much if you do some test cases, this will improve the system

closed 2024-12-12 by dsn27