FriendsOfCADability/CADability #66: Export step problem

happydpc, 2022-01-07:
### Export step fail
After ruling the surface, the export step file failed at Serialize the Edge with cur3d is "Path". Because path not impl IExportStep interface.

Here's the testcase file.
[test_ruled.cdb.json.zip](https://github.com/SOFAgh/CADability/files/7830904/test_ruled.cdb.json.zip)

[image]

dsn27, 2022-10-26:
This error happens because CADability.GeoObject.Path does not implement IExportStep.
Thus the cast (curve3d as IExportStep) will return null.

Either implement the interface or use a safe cast.

https://github.com/SOFAgh/CADability/blob/24181919cb5678058b73b3900441a07a826c7ac9/CADability/Edge.cs#L3567-L3575

System.NullReferenceException
  HResult=0x80004003
  Message= Object reference not set to an instance of an object
  Source=CADability
  StackTrace:
   at CADability.Edge.CADability.IExportStep.Export(ExportStep export, Boolean topLevel) in C:\Development\Repos\SOFAgh\CADability\CADability\Edge.cs:line 3572

x68507, 2023-04-25:
It looks like my file containing only a circular surface might have the same issue.  

CADability can import correctly but it cannot export the surface as a STEP file.  The export fails and programs like _NIST STEP File Analyzer and Viewer_ cannot even parse the exported STEP file.  It looks like the **EDGE_LOOP** has no reference entity and the **B_SPLINE_CURVE_WITH_KNOTS** & corresponding **CARTESIAN_POINT** are never written to the file.  

[fma_02-good.zip](https://github.com/SOFAgh/CADability/files/11327161/fma_02-good.zip)

closed 2026-10-09 by dsn27