FriendsOfCADability/CADability #82: RecalcDisplayLists is not called

dsn27, 2022-06-24:
After working some time with Cadability the Display lists are not updated anymore.
In the screenshots you can see that holes become rectangular instead of round and the number 0 is missing from the yellow text.

Most of the time it helps to zoom in/out very far to trigger an update, but this doesn't seem to work here.
But it helps to add a new GeoObject to force an update.

After some digging I could find out that here
https://github.com/SOFAgh/CADability/blob/8497ecf49d67b4a26315dc02dde2fd2c6bbd539c/CADability/ProjectedModel.cs#L1303-L1307
the dirty flag is always false if this error occurs.
Only after adding a new GeoObject this flag is set to true and the model will update.

Any help would be appreciated.

Incorrect display:
[image]
Missing zero:
[image]
After adding a new line:
[image]

dsn27, 2022-06-27:
Steps to reproduce:

1. Draw a circle
2. Zoom in and out very far
3. Zoom back to a normal view
4. Draw a new circle next to the other one
5. Zoom in again
6. The resolution of the new circle is much lower then the old one.
[image]

7. Draw a new line to force the update

SOFAgh, 2022-07-14:
The problem was: when zooming in, the display precision became smaller than the displayPrecision of the current displaylists. So a recalc of the displaylists was made. Now zooming back out and drawing a circle made the circle display with a precision according to the current visible recatangle, which means only a few lines. Zooming in again did not trigger a new recalculation of the displaylists, because they were marked with a high precision. Now a new drawn circle will be displayed with the precision of the display lists. The drawback might be that if you zoom in very far and zoom out again, every following display will be with high precision

closed 2022-07-14 by SOFAgh