FriendsOfCADability/CADability #100: NullReferenceException in DxfReader.ReadLayerStateProperties

Steinblock, 2022-10-12:
Some files cause a `NullReferenceException` because `GetObjectByHandle` returns null.

Example: [linetype_issue.zip](https://github.com/SOFAgh/CADability/files/9765378/linetype_issue.zip)

```
System.NullReferenceException
  HResult=0x80004003
  Message=Object reference not set to an instance of an object.
  Source=CADability.Core
  StackTrace:
   at netDxf.IO.DxfReader.ReadLayerStateProperties(String name, IEnumerator`1 enumerator) in ...\CADability\netDxf\IO\DxfReader.cs:line 11105
```

PR to solve this will follow

dsn27, 2022-10-26:
Hi @Steinblock

I've updated the netDxf library to the latest version. 
This should include the changes you backported from the original source code.

After a short test with the DXF file you attached everything seems to work fine.

Can we close this issue?

Steinblock, 2022-10-26:
@dsn27 

Yes. I just checked it and all my dxf tests (~110 files) have no issues with the current version.

closed 2022-10-26 by Steinblock