FriendsOfCADability/CADability #116: GetSmallestEnclosingRectangle() does not return the smallest rectangle

dsn27, 2022-11-02:
The function GetSmallestEnclosingRectangle() does not always return the smallest rectangle.

This triangle shaped form is 178.227 x 95.001 = 16,931.74² (Green rectangle)
The function returns 201.85 x 89.11 = 17,986.85² (Blue rectangle)

Is there anything we can do to improve the calculation?

[triangle.zip](https://github.com/SOFAgh/CADability/files/9921203/triangle.zip)
[image]

SOFAgh, 2022-11-02:
GetSmallestEnclosingRectangle() is pretty simple: it uses the convex hull of the border, then it takes each line as a basis direction and calculates the extent. It should be simple to debug.

dsn27, 2022-11-07:
Close #119

closed 2022-11-07 by dsn27