FriendsOfCADability/CADability #128: Importing a DXF File fails with CADability.PlaneException

dsn27, 2022-12-16:
This DXF will cause an CADability.PlaneException
[Bemassung.zip](https://github.com/SOFAgh/CADability/files/10243331/Bemassung.zip)

```
CADability.PlaneException
  HResult=0x80131600
  Message=Fehler in der Anwendung.
  Source=CADability
  StackTrace:
   at CADability.Plane..ctor(GeoPoint Location, GeoPoint p1, GeoPoint p2) in C:\Development\Repos\SOFAgh\CADability\CADability\Plane.cs:line 146
   at CADability.DXF.Import.CreateMesh(Mesh mesh) in C:\Development\Repos\SOFAgh\CADability\CADability\ImportDxf.cs:line 1114
   at CADability.DXF.Import.GeoObjectFromEntity(EntityObject item) in C:\Development\Repos\SOFAgh\CADability\CADability\ImportDxf.cs:line 147
   at CADability.DXF.Import.FindBlock(Block entity) in C:\Development\Repos\SOFAgh\CADability\CADability\ImportDxf.cs:line 333
   at CADability.DXF.Import.CreateInsert(Insert insert) in C:\Development\Repos\SOFAgh\CADability\CADability\ImportDxf.cs:line 796
   at CADability.DXF.Import.GeoObjectFromEntity(EntityObject item) in C:\Development\Repos\SOFAgh\CADability\CADability\ImportDxf.cs:line 138
   at CADability.DXF.Import.FillModelSpace(Model model) in C:\Development\Repos\SOFAgh\CADability\CADability\ImportDxf.cs:line 57
   at CADability.DXF.Import.CreateProject() in C:\Development\Repos\SOFAgh\CADability\CADability\ImportDxf.cs:line 104
   at CADability.DXF.Import.get_Project() in C:\Development\Repos\SOFAgh\CADability\CADability\ImportDxf.cs:line 75
   at CADability.Project.ImportDXF(String filename) in C:\Development\Repos\SOFAgh\CADability\CADability\Project.cs:line 1794
   at CADability.Project.ReadFromFile(String FileName, String Format, Boolean useProgress, Boolean makeCompounds) in C:\Development\Repos\SOFAgh\CADability\CADability\Project.cs:line 1823
   at CADability.Project.ReadFromFile(String FileName, String Format) in C:\Development\Repos\SOFAgh\CADability\CADability\Project.cs:line 1854
   at CADability.FrameImpl.OnFileOpen(String fileName) in C:\Development\Repos\SOFAgh\CADability\CADability\Frame.cs:line 2335
   at CADability.FrameImpl.OnCommand(String MenuId) in C:\Development\Repos\SOFAgh\CADability\CADability\Frame.cs:line 997
   at CADability.Forms.CadFrame.OnCommand(String MenuId) in C:\Development\Repos\SOFAgh\CADability\CADability.Forms\CadFrame.cs:line 71
   at CADability.Forms.ToolBars.TagInfo.ButtonClicked(Object sender, EventArgs e) in C:\Development\Repos\SOFAgh\CADability\CADability.Forms\ToolBars.cs:line 94
   at System.Windows.Forms.ToolStripItem.RaiseEvent(Object key, EventArgs e)
   at System.Windows.Forms.ToolStripButton.OnClick(EventArgs e)
   at System.Windows.Forms.ToolStripItem.HandleClick(EventArgs e)
   at System.Windows.Forms.ToolStripItem.HandleMouseUp(MouseEventArgs e)
   at System.Windows.Forms.ToolStripItem.FireEventInteractive(EventArgs e, ToolStripItemEventType met)
   at System.Windows.Forms.ToolStripItem.FireEvent(EventArgs e, ToolStripItemEventType met)
   at System.Windows.Forms.ToolStrip.OnMouseUp(MouseEventArgs mea)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.ToolStrip.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.DebuggableCallback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)
   at System.Windows.Forms.UnsafeNativeMethods.DispatchMessageW(MSG& msg)
   at System.Windows.Forms.Application.ComponentManager.System.Windows.Forms.UnsafeNativeMethods.IMsoComponentManager.FPushMessageLoop(IntPtr dwComponentID, Int32 reason, Int32 pvLoopData)
   at System.Windows.Forms.Application.ThreadContext.RunMessageLoopInner(Int32 reason, ApplicationContext context)
   at System.Windows.Forms.Application.ThreadContext.RunMessageLoop(Int32 reason, ApplicationContext context)
   at System.Windows.Forms.Application.Run(Form mainForm)
   at CADability.App.Program.Main(String[] args) in C:\Development\Repos\SOFAgh\CADability\CADability.App\Program.cs:line 19

```

SOFAgh, 2022-12-16:
This file doesn't cause an exception. CreateMesh is not called. Maybe it is the wrong file?

dsn27, 2022-12-16:
You are right. This file will not cause an exception.
But it can not be read by netDxf because the DXF Version is AutoCad12.

So it has to go through the conversion process and it seems that your converter is producing a different result than mine.

Attached is the converted file. Which will cause the error.
But now I'm not sure wether this is a problem of CADability or the converter that I'm using.
[Bemassung_Konvertiert.zip](https://github.com/SOFAgh/CADability/files/10244697/Bemassung_Konvertiert.zip)

SOFAgh, 2022-12-16:
Fixed the bug in my branch "NextStep" Will merge into master after more tests.
The bug was caused by a mesh containing a triangle with two identical vertices.

dsn27, 2022-12-20:
Thank you for your help.
I reimplemented the fix in the master branch. The try/catch was changed a little bit, to catch only the PlaneException that was raised.

Wouldn't it be better to remove idectical points instead of ignoring the error?

Unfortunately I pushed the fix to the master branch instead of createing a new branch.
Sorry for that.

closed 2023-04-28 by dsn27