FriendsOfCADability/CADability #129: DXF Text is missplaced after export

Steinblock, 2023-01-03:
I have a simple DXF file with a text object (the yellow one)

[image]

If I export the file as DXF the text is at the wrong position (zero) and has a wrong orientation.

[image]

Files:  
[original.zip](https://github.com/SOFAgh/CADability/files/10337362/original.zip)  
[after_export.zip](https://github.com/SOFAgh/CADability/files/10337364/after_export.zip)

Steinblock, 2023-01-03:
So the reason that happens is because in `ExportDxf.cs` the line `res.TransformBy(Matrix4(toText)); ` is commented out. 

```
        private netDxf.Entities.Text ExportText(GeoObject.Text text)
        {
            System.Drawing.FontStyle fs = System.Drawing.FontStyle.Regular;
            if (text.Bold) fs |= System.Drawing.FontStyle.Bold;
            if (text.Italic) fs |= System.Drawing.FontStyle.Italic;
            System.Drawing.Font font = new System.Drawing.Font(text.Font, 1000.0f, fs);
            netDxf.Entities.Text res = new netDxf.Entities.Text(text.TextString, Vector2.Zero, text.TextSize * 1000 / font.Height, new TextStyle(text.Font + ".ttf"));
            ModOp toText = ModOp.Fit(GeoPoint.Origin, new GeoVector[] { GeoVector.XAxis, GeoVector.YAxis, GeoVector.ZAxis }, text.Location, new GeoVector[] { text.LineDirection.Normalized, text.GlyphDirection.Normalized, text.LineDirection.Normalized ^ text.GlyphDirection.Normalized });
            // res.TransformBy(Matrix4(toText)); // easier than setting normal and rotation
            return res;
        }
```

https://github.com/SOFAgh/CADability/blob/82ffb3400f8971b8fb756c8e96a28f9019930eb0/CADability/ExportDxf.cs#L157

If I uncomment it again (and the used Matrix4 method), it works as intended. Is there a reason why this line has been commented out?

SOFAgh, 2023-01-05:
We should probably uncomment this line. I tested with a few files, maybe you have some more files to test with.

Steinblock, 2023-01-06:
I wrote a simple unit test plus fix and created a PR #130

Since even that simple example fails without the change I would assume this solution should be merged.

closed 2023-01-06 by Steinblock