FriendsOfCADability/CADability #131: StackOverflowException on STL import

Syntaxrabbit, 2023-01-16:
Trying to import the attached two STL files leads to a StackOverflowException.
One is smaller than 1MB, the other one is about 10MB.
Both are in human readable format (no binaries).

[9767e0e1-de99-4479-b2c5-9f7f1371c6cf_2023-01-16_14-22-45.zip](https://github.com/SOFAgh/CADability/files/10426268/9767e0e1-de99-4479-b2c5-9f7f1371c6cf_2023-01-16_14-22-45.zip)
[418b1e03-f023-4edf-81aa-860e0ad5f8d3_2023-01-16_14-22-35.zip](https://github.com/SOFAgh/CADability/files/10426269/418b1e03-f023-4edf-81aa-860e0ad5f8d3_2023-01-16_14-22-35.zip)

SOFAgh, 2023-01-16:
This is probably happening in the DEBUG version. There is a code fragment, which tries to "reverse engineer" the stl file, to find connected triangles which describe some of the standard surfaces (plane, cylinder, sphere, cone, torus). The idea is to combine those triangles to simple faces based on these surfaces, to "reverse engineer" the cad data, which was used to produce the stl file. I abandoned this approach, since no one was interested. In the RELEASE version this code is not running and there seems to be no problem.

Syntaxrabbit, 2023-01-16:
Thanks a lot for the clarification. You are right, the release is working fine.
I created a fork for an online cutting tool designer and made some additions for being able to import/export files inside the web browser. (There is some redundant code now.) Feel free to use it or cancel the pull request otherwise.
Have a nice evening!

closed 2023-01-16 by Syntaxrabbit