FriendsOfCADability/CADability #150: just out of the zip build failed error on the rebuild of both solutions in VS2022

DavidWishengrad, 2023-06-06:
Any ideas why?

CADability.sln...
Severity	Code	Description	Project	File	Line	Suppression State
Error	CS2001	Source file 'C:\Users\Dave\source\repos\CADability-master\CADability.Forms\ImportSVG.cs' could not be found.	CADability.Forms	C:\Users\Dave\source\repos\CADability-master\CADability.Forms\CSC	1	Active
Severity	Code	Description	Project	File	Line	Suppression State
Error	CS0006	Metadata file 'C:\Users\Dave\source\repos\CADability-master\CADability.Forms\bin\Debug\CADability.Forms.dll' could not be found	CADability.App	C:\Users\Dave\source\repos\CADability-master\CADability.App\CSC	1	Active
Severity	Code	Description	Project	File	Line	Suppression State
Error	CS0006	Metadata file 'C:\Users\Dave\source\repos\CADability-master\CADability.Forms\bin\Debug\CADability.Forms.dll' could not be found	CADability.App	C:\Users\Dave\source\repos\CADability-master\CADability.App\CSC	1	Active
Severity	Code	Description	Project	File	Line	Suppression State
Error	CS0006	Metadata file 'C:\Users\Dave\source\repos\CADability-master\CADability.Forms\bin\Debug\CADability.Forms.dll' could not be found	CADability.Tests	C:\Users\Dave\source\repos\CADability-master\tests\CADability.Tests\CSC	1	Active

CADability.Forms.sln...
Severity	Code	Description	Project	File	Line	Suppression State
Error	CS2001	Source file 'C:\Users\Dave\source\repos\CADability-master\CADability.Forms\ImportSVG.cs' could not be found.	CADability.Forms	C:\Users\Dave\source\repos\CADability-master\CADability.Forms\CSC	1	Active

Is there something I am missing, an understanding, an order, files, or all of the above or something else too?

I would really like to try this out. I use the netDXF library successfully and was hope this would be another nice help.

Thank you for any help you can provide.

DavidWishengrad, 2023-06-07:
Hi everyone,

I got something running by deleting the svg.cs file reference in the forms project and it built. These other recommendations did not seem to apply.

"In the Cadability solution, remove the error file ImportSVG.cs (it doesn't exist in the solution) and it should build. The other error is you will need to fix the pathing in the DebuggerVisualizers->Properties->Build Events tab.
The other solution is included in the main Cadabilty solution ;)"

I do not remember naming any other changes, but it won't build again for me atm, but I did get to take a look at what I wanted to. 

This program looks very nice. Nice work!

[image]

SOFAgh, 2023-06-12:
Hello David,
this was certainly my fault, with ImportSVG.cs. I was playing around to check what it would need to import svg files. but I think I  excluded it in the project, so I wonder, why it is still referenced bay the project. I opened "CADability.Forms.csproj" with a text editor and found no reference to ImpoerSVG. So how did you "remove the error file ImportSVG.cs"?
Thanks, Gerhard

closed 2023-07-17 by dsn27