FriendsOfCADability/CADability #153: ImportStep does not respect unit

Steinblock, 2023-07-18:
I have a simple step file and CADability shows the width and height as 0.38 and 0.8.

[image]

However the part has been constructed in Autodesk Inventor 2023 and saved with unit meter. Other step viewers show the correct size of 380 and 800.

Here is the file for reference.

[simple-part-mm.zip](https://github.com/SOFAgh/CADability/files/12077106/simple-part-mm.zip)

dsn27, 2024-09-10:
It seems like the unit is not read under all circumstances.
While there is code that is trying to figure out the units, this does not seem to work for this file.
e.g.
https://github.com/SOFAgh/CADability/blob/e6794d6730d618307e2a8d8c5f34f6fbe7d97719/CADability/ImportStep.cs#L4757-L4782

https://github.com/SOFAgh/CADability/blob/e6794d6730d618307e2a8d8c5f34f6fbe7d97719/CADability/ImportStep.cs#L4804-L4825

dsn27, 2024-09-12:
Closed by #182

closed 2024-09-12 by dsn27