FriendsOfCADability/CADability #157: Deadlock in BSpline

dsn27, 2023-08-29:
BSpline ICurve.GetPlanarState can cause a deadlock if Model.RecalcGeoObject is running at the same time.

[image][image]

As locking to This is not a good practice this should be fixed first.
But I'm not sure that this will solve the problem.

https://github.com/SOFAgh/CADability/blob/32a5dff18809b32c43002fb8e307838b57d1598e/CADability/BSpline.cs#L2901C9-L2906

Any suggestions are welcome.

dsn27, 2023-08-29:
Thanks for the reply.

> Changing lock(this) and lock(typeof(...)) accross the entire codebase to dedicated lock(varThatIsOnlyUsedHere) will definitly reduce chances of deadlocks.

Within BSpline.cs lock (this) is used 4x.
I guess the original intention was to "protect" the object from being modified while running the lock section.

Would you rather change this so that every lock statement has it's own object? Or do they need to share a common object?

Steinblock, 2023-08-30:
> Would you rather change this so that every lock statement has it's own object?

The best answer for this is "it depends".

From outside it is a good practice to lock the object you are working on (like Model.RecalcGeoObject does) so you can avoid having multiple threads manipulating the same object. That applies if the object is not thread safe.

In general the best way to do locking would be

a) don't look the object from outside and make sure the object itself is thread safe 

or

b) lock the object from outside everytime you use it in a multithreaded way. The object itself doesn't need locking internally (unless it starts it's own threads of cource)

Since CADability uses locking inside most external locks shouldn't be needed but the lock statements should not use `lock(this)` to avoid conflicts with external locks.

Back to the question

> Would you rather change this so that every lock statement has it's own object?

I propably would vote for having fewer lock objects.
If there is not already a specific lock but a `lock(this)` I would replace these with a `lock(lockVar)` that is used by other `lock(this)` statements as well since there are many cases where code that runs inside a lock will call code that uses a lock itself. If they use different lock objects that might lead to some rare bugs where the data is inconsistant or an exception occures.

This example would cause a race condition while using the same lockVar would not.

```
public void MyMethod()
{
    lock(lockVar1)
    {
        if (this.isInitialized)
        {
             Initialize();
         }
    }
}

public void MyMethods2()
{
    lock(lockVar2)
    {
        if (this.isInitialized)
        {
             Initialize();
         }
    }
}

public void Initialize()
{
    lock(lockVar2)
    {
       this.dict.Add("key", "value");
    }
}
```

On the other hand you could get better performance with more specific locks but that would require some deep analysis of each lock statement so I think the shared `lockVar` with a possible performance tradeoff would be the better approach for now.

TLDR:

Replacing `lock(this)` with `lock(lockVar)` globally and adding a `lockVar` instance to each object would be easy to implement.

I can create a PR for this this since I have a lot of unit tests in my internal project that run in parallel so I would directly notice any issues.

SOFAgh, 2023-08-30:
I think in BSpline we already have two objects for the locking mechanism: "lockApproximationRecalc" and "planeLock". When in `PlanarState ICurve.GetPlanarState() { lock (this)` we change "this" into "planeLock" the special case would be solved (and the intention of the locking in GetPlanarState is exactly to avoid multiple recalculation of the plane).
 Actually looking at this (old) code, I think we should change GetPlane() into
 ```
           if (plane.HasValue) return plane.Value;
           lock (planeLock)
            {
                double MaxDist;
                bool isLinear;
                Plane res = Plane.FromPoints(poles, out MaxDist, out isLinear);
                if (MaxDist < Precision.eps) 
                {
                   plane = res;
                   return res;
                }
 
```
But I don't have time to test this at the moment.

dsn27, 2023-08-31:
Thank you for taking the time to look into this.

I created a commit d7eb93c that includes the suggested changes.
The other lock (this) statements are replaced by lock (bSplineLock) for now.

Moving `if (plane.HasValue) return plane.Value;` to the outside of the lock is a good idea.
But you still have to performe the check inside the lock too. Because the value could change between the check and the release of the lock.
I found an interesting site about this topic here: 
https://www.pluralsight.com/guides/lock-statement-best-practices

I would appreciate if @Steinblock you could run your tests with this commit.

dsn27, 2023-11-22:
Closed #161

closed 2023-11-22 by dsn27