FriendsOfCADability/CADability #158: netDXF to CADability object helpers or examples?

neilclemons, 2023-08-31:
I can use netDXF to import a DXF file. I can use CADability to import a DXF into a Project object but it still apparently comes through as netDXF objects in the project?

I've pored through the documentation and even some of the source code looking for this and I can't find it. Are there any helpers or examples for converting netDXF data into CADability? Do I have to loop through each entity and do it manually?

Steinblock, 2023-08-31:
It's as simple as

```
var project = Project.ReadFromFile("file.dxf", "dxf");
```
That's the code: https://github.com/SOFAgh/CADability/blob/master/CADability/Project.cs#L1809C9-L1809C120

Even if the names are similar, the objects in project are not the same objects as nextDXF uses.

I.E. netDxf has a [netDxf.Entities.Line](https://github.com/SOFAgh/CADability/blob/master/CADability/netDxf/Entities/Line.cs) and CADability has a [CADability.GeoObject.Line](https://github.com/SOFAgh/CADability/blob/master/CADability/Line.cs)

The conversion happens in `ImportDxf.cs`, see [private IGeoObject GeoObjectFromEntity(EntityObject item)](https://github.com/SOFAgh/CADability/blob/master/CADability/ImportDxf.cs#L125)

```
            switch (item)
            {
                case netDxf.Entities.Line dxfLine: res = CreateLine(dxfLine); break;
                case netDxf.Entities.Ray dxfRay: res = CreateRay(dxfRay); break;
                case netDxf.Entities.Arc dxfArc: res = CreateArc(dxfArc); break;
                case netDxf.Entities.Circle dxfCircle: res = CreateCircle(dxfCircle); break;
                case netDxf.Entities.Ellipse dxfEllipse: res = CreateEllipse(dxfEllipse); break;
                case netDxf.Entities.Spline dxfSpline: res = CreateSpline(dxfSpline); break;
                case netDxf.Entities.Face3D dxfFace: res = CreateFace(dxfFace); break;
                case netDxf.Entities.PolyfaceMesh dxfPolyfaceMesh: res = CreatePolyfaceMesh(dxfPolyfaceMesh); break;
                case netDxf.Entities.Hatch dxfHatch: res = CreateHatch(dxfHatch); break;
                case netDxf.Entities.Solid dxfSolid: res = CreateSolid(dxfSolid); break;
                case netDxf.Entities.Insert dxfInsert: res = CreateInsert(dxfInsert); break;
                case netDxf.Entities.Polyline2D dxfPolyline2D: res = CreatePolyline2D(dxfPolyline2D); break;
                case netDxf.Entities.MLine dxfMLine: res = CreateMLine(dxfMLine); break;
                case netDxf.Entities.Text dxfText: res = CreateText(dxfText); break;
                case netDxf.Entities.Dimension dxfDimension: res = CreateDimension(dxfDimension); break;
                case netDxf.Entities.MText dxfMText: res = CreateMText(dxfMText); break;
                case netDxf.Entities.Leader dxfLeader: res = CreateLeader(dxfLeader); break;
                case netDxf.Entities.Polyline3D dxfPolyline3D: res = CreatePolyline3D(dxfPolyline3D); break;
                case netDxf.Entities.Point dxfPoint: res = CreatePoint(dxfPoint); break;
                case netDxf.Entities.Mesh dxfMesh: res = CreateMesh(dxfMesh); break;
                default:
                    System.Diagnostics.Trace.WriteLine("dxf: not imported: " + item.ToString());
                    break;
            }
```

neilclemons, 2023-08-31:
> It's as simple as
>
> ```
> var project = Project.ReadFromFile("file.dxf", "dxf");
> ```
>
> That's the code: https://github.com/SOFAgh/CADability/blob/master/CADability/Project.cs#L1809C9-L1809C120
>
> Even if [...]

Thank you. That sets me down the path I need to finish my importer. The documentation is confusing. The ReadFromFile() method doesn't mention file types and the CADability.DXF.Import() method specifically says it converts the file to netDXF objects. Navigating the source is a bit tricky as the documentation is full of broken links.

closed 2023-09-05 by dsn27