FriendsOfCADability/CADability #163: Extruding a path containing an ellipse along  path also containing an ellipse doesn't work.

niempi2, 2023-12-22:
The following piece of code is supposed to extrude a shape along a curved path. The straight part of the shape is extruded, but the curved part is not. 

var project = Project.CreateSimpleProject();

CADability.GeoObject.Path path = Path.Construct();
path.Add(Line.MakeLine(new GeoPoint(0, 0), new GeoPoint(0, 100)));
var ellipse = Ellipse.Construct();
ellipse.SetArc3Points(new GeoPoint(0, 100), new GeoPoint(50, 50), new GeoPoint(0, 0),Plane.XYPlane);
path.Add(ellipse);
var extrudePath = Path.Construct();

var extEllipse = Ellipse.Construct();
extEllipse.SetArc3Points(new GeoPoint(0,0), new GeoPoint(0, 0, 100),new GeoPoint(0,60,50),Plane.YZPlane);
extrudePath.Add(extEllipse);
var pipe = Make3D.MakePipe(path, extrudePath, project);
project.GetActiveModel().Add(pipe);
project.ExportToSTL("c:/temp/e.stl");

SOFAgh, 2023-12-27:
Currently extruding ellipses along another ellipse is only implemented for the perpendicular case, i.e. when the extrusion starts  perpendicular to the plane of the ellipse to be extruded (and results in a torus). If this is not the case, the extruded surface might be self intersection or have a strange shape. It could be implemented as a SurfaceOfRevolution , but I think you will rarely find such surfaces in real models. 
[image]

niempi2, 2023-12-28:
Thank you for the answer. You might consider throwing a notimplementedexception in this case.

closed 2023-12-28 by niempi2