FriendsOfCADability/CADability #185: Exception during design of CadCanvas control

johnnyontheweb, 2024-10-31:
Hello,
I'm trying to using CadCanvas control in a Winform app. During design, devEnv.exe is often terminated by this error:
Cannot find method 'Void CADability.IPaintTo3D.PrepareIcon(System.Drawing.Bitmap)'. 
System.MissingMethodException

stacktrace:
   in CADability.Forms.CadCanvas.Dispose(Boolean disposing)
   in System.ComponentModel.Component.Finalize() in System.ComponentModel\Component.cs: riga 85

I discovered this with another instance of devEnv attached for debugging the main one.
Any clue?

johnnyontheweb, 2024-12-05:
Hello, thanks for replying. It's a simple Winform project with a CadControl in it. The code to initialize the view is:
```
        Dim cadFrame = New CadFrame(CadCanvas1, Me)
        Dim proj = cadFrame.Project
        Dim newGDIView As New GDI2DView(proj, proj.GetActiveModel, cadFrame)
        proj.GdiViews.Add(newGDIView)
        cadFrame.ActiveView = newGDIView
```
I believe that's all. 
I often get the exception reported (System.MissingMethodException) expecially when opening the form in Visual Studio form designer.

dsn27, 2024-12-13:
After some trying I can now reproduce this error.
Mine is a little bit different:

I just need to open MainForm.cs in the Visual Studio designer.
[image]

Visual Studio sometimes crashes.
Eventlog:

```
System.MissingMethodException
   at CADability.Forms.CadCanvas.Dispose(Boolean)
   at System.ComponentModel.Component.Finalize()
```

This doesn't happen if you open the CadForm from CADability.Forms.

I used Fuslogvw to try to get more information:
```

=== Pre-bind state information ===
LOG: DisplayName = CADability.Forms, Version=1.0.0.0, Culture=neutral, PublicKeyToken=62ba9c16fd2eecf7
 (Fully-specified)
LOG: Appbase = file:///C:/Program Files/Microsoft Visual Studio/2022/Professional/Common7/IDE/
LOG: Initial PrivatePath = NULL
LOG: Dynamic Base = NULL
LOG: Cache Base = NULL
LOG: AppName = devenv.exe
Calling assembly : (Unknown).
===
LOG: This bind starts in default load context.
LOG: Using application configuration file: C:\Users\UserName\AppData\Local\Microsoft\VisualStudio\17.0_429554e8\devenv.exe.config
LOG: Using host configuration file: 
LOG: Using machine configuration file from C:\Windows\Microsoft.NET\Framework64\v4.0.30319\config\machine.config.
LOG: Post-policy reference: CADability.Forms, Version=1.0.0.0, Culture=neutral, PublicKeyToken=62ba9c16fd2eecf7
LOG: GAC Lookup was unsuccessful.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Professional/Common7/IDE/CADability.Forms.DLL.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Professional/Common7/IDE/CADability.Forms/CADability.Forms.DLL.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Professional/Common7/IDE/PublicAssemblies/CADability.Forms.DLL.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Professional/Common7/IDE/PublicAssemblies/CADability.Forms/CADability.Forms.DLL.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Professional/Common7/IDE/PrivateAssemblies/CADability.Forms.DLL.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Professional/Common7/IDE/PrivateAssemblies/CADability.Forms/CADability.Forms.DLL.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Professional/Common7/IDE/CommonExtensions/Microsoft/TestWindow/CADability.Forms.DLL.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Professional/Common7/IDE/CommonExtensions/Microsoft/TestWindow/CADability.Forms/CADability.Forms.DLL.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Professional/Common7/IDE/CommonExtensions/Platform/Debugger/CADability.Forms.DLL.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Professional/Common7/IDE/CommonExtensions/Platform/Debugger/CADability.Forms/CADability.Forms.DLL.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Professional/Common7/IDE/PrivateAssemblies/DataCollectors/CADability.Forms.DLL.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Professional/Common7/IDE/PrivateAssemblies/DataCollectors/CADability.Forms/CADability.Forms.DLL.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Professional/Common7/IDE/PrivateAssemblies/DataCollectors/x64/CADability.Forms.DLL.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Professional/Common7/IDE/PrivateAssemblies/DataCollectors/x64/CADability.Forms/CADability.Forms.DLL.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Professional/Common7/IDE/CADability.Forms.EXE.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Professional/Common7/IDE/CADability.Forms/CADability.Forms.EXE.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Professional/Common7/IDE/PublicAssemblies/CADability.Forms.EXE.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Professional/Common7/IDE/PublicAssemblies/CADability.Forms/CADability.Forms.EXE.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Professional/Common7/IDE/PrivateAssemblies/CADability.Forms.EXE.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Professional/Common7/IDE/PrivateAssemblies/CADability.Forms/CADability.Forms.EXE.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Professional/Common7/IDE/CommonExtensions/Microsoft/TestWindow/CADability.Forms.EXE.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Professional/Common7/IDE/CommonExtensions/Microsoft/TestWindow/CADability.Forms/CADability.Forms.EXE.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Professional/Common7/IDE/CommonExtensions/Platform/Debugger/CADability.Forms.EXE.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Professional/Common7/IDE/CommonExtensions/Platform/Debugger/CADability.Forms/CADability.Forms.EXE.
[... cut ...]
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Professional/Common7/IDE/PrivateAssemblies/DataCollectors/x64/CADability.Forms.EXE.
LOG: Attempting download of new URL file:///C:/Program Files/Microsoft Visual Studio/2022/Professional/Common7/IDE/PrivateAssemblies/DataCollectors/x64/CADability.Forms/CADability.Forms.EXE.
LOG: All probing URLs attempted and failed.

```

I'm not sure why Visual Studio is searching in these places for the DLL. But even installing CADability.Forms.dll and CADability.dll into the GAG does not help.

Any ideas would be welcome.

dsn27, 2026-10-09:
Closing this as stale. The setup it was reported against no longer exists.

At the time, CADability.Forms targeted `net48` and ran inside Visual Studio's in-process .NET Framework designer. CADability.dll is netstandard2.0 and uses `System.Drawing.Common`. A `MissingMethodException` for `IPaintTo3D.PrepareIcon(System.Drawing.Bitmap)` fits a `Bitmap` type mismatch between those two worlds.

Since commit 517d643f ("Upgrade CADability.Forms, CADability.App, and CADability.Tests to .NET 8"), CADability.Forms targets `net8.0-windows`. Visual Studio's designer then runs out of process, so that mismatch should no longer occur. We could not re-test the designer itself, though, so this is not a confirmed fix.

If you still get the exception with the current version, please reopen and include the stack trace.

---
_Generated by [Claude Code](https://claude.ai/code)_

closed 2026-10-09 by dsn27