FriendsOfCADability/CADability #187: DXF spline import tied to frame settings. Cannot be used headless.

ChrisClems, 2024-12-04:
On line 533-534, ImportDxf is attempting to get the precision value to approximate a spline: https://github.com/FriendsOfCADability/CADability/blob/ddb68c06672904949ac7961aba92c6ddff4078fa/CADability/ImportDxf.cs#L533

It is using Project.Frame.GetDoubleSetting(). In a headless environment, Project.Frame will be null so it is throwing an exception and failing to import the DXF when a spline that triggers this line of code is encountered.

I tried bypassing this in an attempt to troubleshoot by hard-coding my maxError to the default of 0.01. Somehow, when I do this, the program breaks after the constructor as `doc` will now be null. I've set up breakpoints all over to try to track down why this is happening, but I can't figure out why commenting out this line would break the Import constructor which runs far before the CreateSpline() method is ever called. I set a breakpoint at the top of `GeoObjectFromEntity()` to confirm it's never reaching any code past the project init.

This is the entirety of the program I'm using to test this:

```C#
var cadabDxf = new CADability.DXF.Import(@"E:\126582-01-old.dxf").Project;

Console.WriteLine(cadabDxf.ToString());
```

The only modification I've made is to maxError in CreateSpline():

```C#
private IGeoObject CreateSpline(netDxf.Entities.Spline spline)
        {
            int degree = spline.Degree;
            if (spline.ControlPoints.Length == 0 && spline.FitPoints.Count > 0)
            {
                BSpline bsp = BSpline.Construct();
                GeoPoint[] fp = new GeoPoint[spline.FitPoints.Count];
                for (int i = 0; i < fp.Length; i++)
                {
                    fp[i] = GeoPoint(spline.FitPoints[i]);
                }
                bsp.ThroughPoints(fp, spline.Degree, spline.IsClosed);
                return bsp;
            }
            else
            {
                bool forcePolyline2D = false;
                GeoPoint[] poles = new GeoPoint[spline.ControlPoints.Length];
                double[] weights = new double[spline.ControlPoints.Length];
                for (int i = 0; i < poles.Length; i++)
                {
                    poles[i] = GeoPoint(spline.ControlPoints[i]);
                    weights[i] = spline.Weights[i];

                    if (i > 0 && (poles[i] | poles[i - 1]) < Precision.eps)
                    {
                        forcePolyline2D = true;
                    }
                }
                double[] kn = new double[spline.Knots.Length];
                for (int i = 0; i < kn.Length; ++i)
                {
                    kn[i] = spline.Knots[i];
                }
                if (poles.Length == 2 && degree > 1)
                {   // damit geht kein vernünftiger Spline, höchstens mit degree=1
                    GeoObject.Line l = GeoObject.Line.Construct();
                    l.StartPoint = poles[0];
                    l.EndPoint = poles[1];
                    return l;
                }
                BSpline bsp = BSpline.Construct();
                //TODO: Can Periodic spline be not closed?
                if (bsp.SetData(degree, poles, weights, kn, null, spline.IsClosedPeriodic))
                {
                    // BSplines with inner knots of multiplicity degree+1 make problems, because the spline have no derivative at these points
                    // so we split these splines
                    List<int> splitKnots = new List<int>();
                    for (int i = degree + 1; i < kn.Length - degree - 1; i++)
                    {
                        if (kn[i] == kn[i - 1])
                        {
                            bool sameKnot = true;
                            for (int j = 0; j < degree; j++)
                            {
                                if (kn[i - 1] != kn[i + j]) sameKnot = false;
                            }
                            if (sameKnot) splitKnots.Add(i - 1);
                        }
                    }
                    if (splitKnots.Count > 0)
                    {
                        List<ICurve> parts = new List<ICurve>();
                        BSpline part = bsp.TrimParam(kn[0], kn[splitKnots[0]]);
                        if (CADability.GeoPoint.Distance(part.Poles) > Precision.eps && (part as ICurve).Length > Precision.eps) parts.Add(part);
                        for (int i = 1; i < splitKnots.Count; i++)
                        {
                            part = bsp.TrimParam(kn[splitKnots[i - 1]], kn[splitKnots[i]]);
                            if (CADability.GeoPoint.Distance(part.Poles) > Precision.eps && (part as ICurve).Length > Precision.eps) parts.Add(part);
                        }
                        part = bsp.TrimParam(kn[splitKnots[splitKnots.Count - 1]], kn[kn.Length - 1]);

                        if (CADability.GeoPoint.Distance(part.Poles) > Precision.eps && (part as ICurve).Length > Precision.eps) parts.Add(part);
                        GeoObject.Path path = GeoObject.Path.Construct();
                        path.Set(parts.ToArray());
                        return path;
                    }
                    // if (spline.IsPeriodic) bsp.IsClosed = true; // to remove strange behavior in hünfeld.dxf

                    if (forcePolyline2D)
                    {
                        //Look at https://github.com/SOFAgh/CADability/issues/173 to see why this is done.

                        ICurve curve = (ICurve)bsp;
                        //Use approximate to get the count of lines that will be needed to convert the spline into a Polyline2D
                        //double maxError = project.Frame.GetDoubleSetting("Approximate.Precision", 0.01);
                        double maxError = 0.01;
                        ICurve approxCurve = curve.Approximate(true, maxError);

                        int usedCurves = 0;
                        if (approxCurve is GeoObject.Line)
                            usedCurves = 2;
                        else
                            usedCurves = approxCurve.SubCurves.Length;

                        netDxf.Entities.Polyline2D p2d = spline.ToPolyline2D(usedCurves);
                        var res = CreatePolyline2D(p2d);
                        
                        return res;
                    }

                    return bsp;
                }
                // strange spline in "bspline-closed-periodic.dxf"
            }
            return null;
        }
```

dsn27, 2024-12-05:
GetDoubleSetting is implemented like this.

```
public double GetDoubleSetting(string Name, double Default)
{
    object s = GetSetting(Name);
    if (s != null)
    {
        try
        {
            return Settings.GetDoubleValue(s);
        }
        catch (InvalidCastException)
        {
            return Default;
        }
    }
    return Default;
}
```

You could create a new function in ImportDxf to help you replace project.Frame.GetDoubleSetting(...).
If this works fine, a PR would be appreciated to fix this use case.

Thanks in advance,
Michel

ChrisClems, 2024-12-05:
> GetDoubleSetting is implemented like this.
>
> ```
> public double GetDoubleSetting(string Name, double Default)
> {
>     object s = GetSetting(Name);
>     if (s != null)
>     {
>         try [...]

I'll play around with in over the next couple of weeks and find a comfortable pattern. I'm doing a lot of work with DXF import and export for a project and I'm hitting a lot of little limitations that I plan to submit PRs for as I have time to test them.

ChrisClems, 2024-12-05:
Turned out to only take a couple of hours. I used the GlobalSettings class to re-use existing functionality without having to change the signature for the DXF export or make it difficult to set on the user side.

In the process I found a bug in SetValue() that wasn't handling DoubleProperties. I added a handler so all DoubleProperties in the GlobalSettings can now be set via code. There might be more missing.

When digging into the GlobalSettings structure, I found more references to Project.Frame for properties. Might be worth opening a new issue just to find them and convert them to library-accessible formats.

PR #188 (2024-12-06): Update spline approximation precision in ImportDxf.cs for headless operation
In implementing this fix, it was found that the SetValue() method was lacking some handlers for certain settings types. Added a statement to set value for settings of type DoubleProperty. There may be more cases that need to be addressed in this method.

This setting can now be control by the user via:

```C#
Settings.GlobalSettings.SetValue("Approximate.Precision", 0.01);
```

Closes #187
closed 2024-12-06 by dsn27