FriendsOfCADability/CADability #192: Fix Visualizer for Visual Studio >17.6

dsn27, 2024-12-10:
Warning CS0618: 'IVisualizerObjectProvider.GetObject()' is obsolete: 'This method uses binary serialization which is no longer supported, please use IVisualizerObjectProvider3.GetObject instead.'

We should take a look at this warning.
Would be nice if the Visualizers would work again in the new versions of Visual Studio.

dsn27, 2024-12-11:
Don't bother with this issue if you don't use Visual Studio.

It's only nice to have and not very important.

dsn27, 2025-03-19:
The way DebuggerVisualizer works has changed in Visual Studio 2022 (17.9+).

If we adopt the new approach, it will break compatibility with all previous versions of Visual Studio 2022 (<17.9).

To avoid compatibility issues, I recommend ignoring the remaining compiler warnings related to DebuggerVisualizer for now and rewriting this from scratch in CADability 2.0.

For more details, refer to Microsoft’s documentation:
🔗 [New Debugger Visualizer Model](https://learn.microsoft.com/en-us/visualstudio/extensibility/visualstudio.extensibility/debugger-visualizer/debugger-visualizers?view=vs-2022)

#49

stefan-tb, 2025-03-19:
Adopting the code for newer versions of VS seems legit. Even if i never missed the visualizer.

dsn27, 2026-10-09:
Closing as not planned. Commit d3db7052 ("Remove DebuggerVisualizers") removed the CADability.DebuggerVisualizers project from master. With it went the CS0618 warning about `IVisualizerObjectProvider.GetObject()`.

To make visualizers work in current Visual Studio versions, they would need to be rewritten on the new VisualStudio.Extensibility visualizer model, as discussed above. That is not planned at the moment. A prototype exists on the `ShapeItProgress` branch (76e05ff0) as a starting point. If the visualizers are brought back, please open a new issue.

---
_Generated by [Claude Code](https://claude.ai/code)_

closed 2026-10-09 by dsn27