FriendsOfCADability/CADability #194: Circular hole UnsplittedOutline is being lost before CompoundShape is returned from CompoundShape.CreateFromList()

ChrisClems, 2024-12-10:
In the attached DXF, when importing and creating a CompoundShape with the CreateFromList method, the UnsplittedOutline property is not making it back to the returned CompoundShape's hole Borders. I've managed to trace it through several methods in the shape building process but so far have been unable to find out where exactly it's being lost.

[TestHoles.zip](https://github.com/user-attachments/files/18087249/TestHoles.zip)

dsn27, 2024-12-11:
Hi Chris.

It would be really helpfull if you could provide a complete sample that is "ready" to run.

I tried it like this:
```

private void button1_Click(object sender, EventArgs e)
{
    var m = cadFrame.Project.GetActiveModel();
    var geos = m.AllObjects;
    var cs = CompoundShape.CreateFromList(geos, 0.1, out Plane plane, true);
}
```

To find out why UnsplittedOutline is null, I have put a break point on the setter of the property.

```
>	CADability.dll!CADability.Shapes.Border.UnsplittedOutline.set(CADability.Curve2D.ICurve2D value) Line 1179	C#
 	CADability.dll!CADability.Shapes.Border.Segments.set(CADability.Curve2D.ICurve2D[] value) Line 760	C#
 	CADability.dll!CADability.Shapes.Border.Reduce(double prec) Line 1082	C#
 	CADability.dll!CADability.Shapes.SimpleShape.Reduce(double precision) Line 1141	C#
 	CADability.dll!CADability.Shapes.CompoundShape.CreateFromList(CADability.GeoObject.GeoObjectList TheObjects, double maxGap, out CADability.Plane plane, bool partInPart) Line 2109	C#
 	CADability.Forms.dll!CADability.Forms.CadForm.button1_Click(object sender, System.EventArgs e) Line 138	C#
 	[External Code]	
 	CADability.App.exe!CADability.App.Program.Main(string[] args) Line 19	C#
```

Turns out the Segments are reset by the Reduce function, which in turn sets the UnsplittedOutline to null.

Now we only have to figure out what to do about it. But I don't have an idea right now.

The Property UnsplittedOutline was added by myself as I needed to retain the original objects like circles etc.
A better solution would be to find a way to not call SplitSingleCurve at all and teach the BorderOperation to deal with this.

ChrisClems, 2024-12-11:
I may be missing something, but do we need to be setting UnsplittedOutline every time the segments are updated at all? If UnsplittedOutline is a preservation of the original geometry, further operations on the segments shouldn't automatically impact the UnsplittedOutline object. If transforms are done on the segments that impact congruence with the original object, then it would be up to the user to also update the UnsplittedOutline property with the same transforms to keep them in sync instead of just erasing the value.

closed 2026-10-09 by dsn27