FriendsOfCADability/CADability #250: GeoVectorException when calling GetSmallestEnclosingRectangle

dsn27, 2025-03-04:
The following exception can occure if you try to run GetSmallestEnclosingRectangle.
```
CADability.GeoVectorException
  HResult=0x80131600
  Message=Fehler in der Anwendung.
  Source=CADability
  StackTrace:
   at CADability.GeoVector2D.get_Normalized() in Z:\Development\Repos\SOFAgh\CADability\GeoPoint.cs:line 1488
   at CADability.Curve2D.GeneralCurve2D.GetTriangulationBasis(GeoPoint2D[]& points, GeoVector2D[]& directions, Double[]& parameters) in Z:\Development\Repos\SOFAgh\CADability\GeneralCurve2D.cs:line 1574
   at CADability.Curve2D.GeneralCurve2D.MakeTriangulation() in Z:\Development\Repos\SOFAgh\CADability\GeneralCurve2D.cs:line 1592
   at CADability.Curve2D.GeneralCurve2D.GetTriangulationPoints(GeoPoint2D[]& basisPoints, GeoPoint2D[]& vertices, GeoVector2D[]& directions, Double[]& parameters) in Z:\Development\Repos\SOFAgh\CADability\GeneralCurve2D.cs:line 1714
   at CADability.Curve2D.Curves2D.TangentLines(ICurve2D first, ICurve2D second) in Z:\Development\Repos\SOFAgh\CADability\Curves2D.cs:line 34
   at CADability.Shapes.Border.ConvexHull() in Z:\Development\Repos\SOFAgh\CADability\Border.cs:line 4345
   at CADability.Shapes.Border.GetSmallestEnclosingRectangle() in Z:\Development\Repos\SOFAgh\CADability\Border.cs:line 4541
   at CADability.Forms.CadForm.button1_Click(Object sender, EventArgs e) in Z:\Development\Repos\SOFAgh\CADability.Forms\CadForm.cs:line 150
   at System.Windows.Forms.Control.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ButtonBase.WndProc(Message& m)
   at System.Windows.Forms.Button.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.DebuggableCallback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)
   at System.Windows.Forms.UnsafeNativeMethods.DispatchMessageW(MSG& msg)
   at System.Windows.Forms.Application.ComponentManager.System.Windows.Forms.UnsafeNativeMethods.IMsoComponentManager.FPushMessageLoop(IntPtr dwComponentID, Int32 reason, Int32 pvLoopData)
   at System.Windows.Forms.Application.ThreadContext.RunMessageLoopInner(Int32 reason, ApplicationContext context)
   at System.Windows.Forms.Application.ThreadContext.RunMessageLoop(Int32 reason, ApplicationContext context)
   at System.Windows.Forms.Application.Run(Form mainForm)
   at CADability.App.Program.Main(String[] args) in Z:\Development\Repos\SOFAgh\CADability.App\Program.cs:line 19

```
DXF File to reproduce this error:
[1.zip](https://github.com/user-attachments/files/19066352/1.zip)

Code to reproduce:

```
private void button1_Click(object sender, EventArgs e)
{
	var geo = cadFrame.ActiveView.Model.AllObjects;
	
	var curve2DData = GeoListToCurve2DArray(geo);

	//Does not work
	var cs = CompoundShape.CreateFromList(curve2DData, 0.01, true, out GeoObjectList deadObjects);

	//Works
	//var cs = CompoundShape.CreateFromList(geo, 0.01, out Plane plane, true);

	var ss = cs.SimpleShapes[0];
	var rect = ss.Outline.GetSmallestEnclosingRectangle();

	var width = rect.Width;
	var height = rect.Height;

}

public static ICurve2D[] GeoListToCurve2DArray(GeoObjectList list)
{
	//Convert ICurve to ICurve2D
	List<ICurve2D> curves2d = new List<ICurve2D>(list.Count);

	foreach (IGeoObject g in list)
	{
		if (g is ICurve c)
		{
			//Project curve to right plane
			ICurve2D curve2D = c.GetProjectedCurve(Plane.XYPlane);

			//Cast curve to GeoObject to save the userdata and attributes
			IGeoObject geoCurve = (IGeoObject)c;
			curve2D.UserData.Add(geoCurve.UserData);
			curves2d.Add(curve2D);
		}
	}

	return curves2d.ToArray();
}
```

SOFAgh, 2025-03-04:
Without actually testinig, I see that in the IGeoObject version the 2d curves are "Reduced". Maybe there are 2d lines with startPoint==endPoint, which cause the GeoVectorException. See source code of CompoundShape.CreateFromList(GeoObjectList

```
                Reduce2D r2d = new Reduce2D();
                r2d.Add((ICurve2D[])curves2D.ToArray(typeof(ICurve2D)));
                r2d.Precision = maxGap;
                r2d.BreakPolylines = true;
                r2d.OutputMode = Reduce2D.Mode.Simple;
                CurveGraph pg = new CurveGraph(r2d.Reduced, maxGap);

```

ChrisClems, 2025-03-04:
I think this is from the arc centered at 1487.82552825424 x 1371.15793803973 ending up with a 0 sweep angle and 0 length. It looks like both the start and end points are set to the end point in the reduced segment.

I know there are some bugs in working with netDXF arc translations related to CW vs CCW arcs and their normals. The arcs in this DXF have -1 Z normals. I suspect this is breaking something in GetProjectedCurve or the reduction. Look at your coordinate systems for your geometry in the two methods. Using the GeoObjectList directly is detecting and returning the plane instead of forcing it and that plane is -1, -1, -1. Forcing an XY plane in the Curve2D conversion is probably breaking all of the "upside down" geometry.

I may be totally off here. I just scratched the surface. Unfortunately, I don't have time to dig in too much further at the moment.

I've put some work into fixing the arc issue in my ACadSharp branch that I've been working on. The intent was hopefully to loop back on it and also fix up the netDXF import/export to better handle inverted normals. I noticed that the geometry imported with AcadSharp will not create a list of curves that will build a compound shape with this DXF.

SOFAgh, 2025-03-05:
CADability should handle arcs on any plane, be it z==-1 or z==1 or any skewed plane correctly. CW or CCW is always in the sense of that plane. Maybe there is a bug in importing this arc from DXF. Unfortunately I cannot open the "1.dxf" file due to a encoding problem in `encoding = Encoding.GetEncoding(codepage);` Maybe you have another version of this dxf file, so I could test.

ChrisClems, 2025-03-05:
> CADability should handle arcs on any plane, be it z==-1 or z==1 or any skewed plane correctly. CW or CCW is always in the sense of that plane. Maybe there is a bug in importing this arc from DXF. [...]

When my schedule slows down, I'll try to remember to loop back on this and provide an example. I have a bunch of DXFs that, when just headlessly importing then exporting, will end up with arcs that are inverted (same center, start and end points swapped so they're going the wrong "direction" in the exported drawing).

That said, @SOFAgh, I believe I was wrong about this. The symptoms are similar to some issues I was dealing with due to arc translation.

At some point in Border.ConvexHull(), the arc centered at the aforementioned coordinates is being reduced to 0-length and passed to Curves2D.TangentLines(). I believe there's a bug somewhere in the outer bulge processing [right around here](https://github.com/FriendsOfCADability/CADability/blob/1fc6727ac14ef46fbefd6bdf28bb75b7f8d47244/CADability/Border.cs#L4295) that's causing this but I haven't been able to drill down the exact step.

ChrisClems, 2025-03-06:
@SOFAgh Are you getting an exception or a message box from the debug assertion? There shouldn't be any encoding issues with that DXF. The DWGCODEPAGE is ANSI_1252. If netDXF has problems reading the encoding it automatically falls back to Windows-1252 which is the same encoding. If you're missing Windows-1252 support there should be an exception being thrown from System, as far as I know.

dsn27, 2025-03-16:
@SOFAgh 
Any luck in reproducing the original GeoVectorException error?
Do you have an idea what could go wrong?

dsn27, 2025-03-20:
@ChrisClems 
Chris, could you write me your email address to mmetzler@hard-coded.cologne ?

dsn27, 2025-03-22:
I've made some progress in debugging this. 

Set a break point at line 2106 in Border.cs

```
else if (j == endIndex)
{
    ICurve2D trimmed = seg.Trim(0.0, endPos);
    if (trimmed != null) part.Add(trimmed);          <---- Set here
}
```

Use this conditional expression for break point: 
`trimmed.GetType().Name == "Arc2D" && trimmed.Sweep == 0`

This reveals that an invalid Arc2D (sweep = 0) is added to the result.
Note: the original segment is valid — the invalidity comes after trimming.

This happens because seg.Trim(0.0, 0.0) is called.
Which will always result in an invalid Arc2D.

So the question is:
👉 Shouldn't Arc2D.Trim() throw an exception for this invalid case?
👉 How to fix SplitCyclical?

What do you think?

dsn27, 2025-03-28:
I modified Arc2D.Trim() so it returns null if the resulting arc would be invalid.

```
public override ICurve2D Trim(double startPos, double endPos)
{
    //If both parameters are 0.0d this will produce an invalid arc
    if (startPos == 0.0d && endPos == 0.0d)
        return null;

    Arc2D res = new Arc2D(Center, Radius, start, sweep);
    Angle st = start + startPos * sweep;
    Angle end = start + endPos * sweep;
    res.sweep = new SweepAngle(st, end, (sweep > 0.0) ^ (startPos > endPos));
    res.start = st;

    //Return null if the trimmed Arc2D would be invalid.
    if(res.sweep.Degree < Precision.eps)
        return null;

    return res;
}
```

After this change, the program runs without any errors.

@SOFAgh Do you think this could have any side effects?

PR #257 (2025-04-01): Fix issue in Arc2D.Trim if startPos and endPos are near 0
Fix #250
closed 2025-04-01 by dsn27