FriendsOfCADability/CADability #253: MultipleChoiceInput & mouse move events

davidebazzi, 2025-03-17:
If an action as a GeoObjectInput and a MultipleChoiceInput, when the GeoObjectInput is selected it reacts to the mouse move events.
If the user selects the MultipleChoiceInput, that can't reacts to the mouse move events, the mouse move events are still handled by the GeoObjectInput.
I find this behaviour wrong.
For me the mouse move events must only be handled by the selected input.

davidebazzi, 2025-03-17:
My proposal is to handle the PropertyEntryChangedStateEvent like other Inputs, for example DoubleInput.

davidebazzi, 2025-03-17:
@SOFAgh what do you think?

dsn27, 2026-10-10:
Fixed in #380 as an opt-in property rather than a changed default.

Subscribing `MultipleChoiceInput`/`BooleanInput` to `PropertyEntryChangedStateEvent` for everyone would change 36 built-in actions that combine these inputs with a mouse input. One example is moving objects while the "copy" option is selected: the move preview would stop until the vector field is selected again. So both inputs now have `CapturesMouse` (default `false`):

```csharp
MultipleChoiceInput mode = new MultipleChoiceInput("MyAction.Mode", "MyAction.Mode.Values");
mode.CapturesMouse = true;
```

With `CapturesMouse = true`, selecting the input makes it the current input of the action, the same as for the other inputs. Mouse moves in the view then go to it: it ignores them, or passes them on to `ForwardMouseInputTo` if you set that. Your `GeoObjectInput` no longer reacts while the choice is selected.

This is new public API, so it ships with the next minor release.

---
_Generated by [Claude Code](https://claude.ai/code)_

closed 2026-10-10 by dsn27