FriendsOfCADability/CADability #254: MultipleChoiceInput with variables choices

davidebazzi, 2025-03-17:
I have the requirement to filter out some of the initial possible choices.
This is not yet possible neither with the [values from the resources](https://github.com/FriendsOfCADability/CADability/blob/d8b0205bfef5c1bdafed78d2a4be047c3071ffc8/CADability/ConstructAction.cs#L3763) neither with the [directValues](https://github.com/FriendsOfCADability/CADability/blob/d8b0205bfef5c1bdafed78d2a4be047c3071ffc8/CADability/ConstructAction.cs#L3764) these values are used during BuildShowProperty and can't be changed during the action use.
To avoid having to implement from scratch an input similar to MultipleChoiceInput, I suggest to allow to instantiate a derived MultipleChoiceProperty in a derived MultipleChoiceInput.
Then in the derived MultipleChoiceProperty it is easy to change the possible choices through [the Choices property](https://github.com/FriendsOfCADability/CADability/blob/d8b0205bfef5c1bdafed78d2a4be047c3071ffc8/CADability/MultipleChoiceProperty.cs#L74)

davidebazzi, 2025-03-17:
@SOFAgh what do you think?

SOFAgh, 2025-03-18:
I didn't check deeply, but in the code it looks like when you change the `Choices `property of the `MultipleChoiceProperty`, next time when you open the dropdown list, there are the modified choices available. `MultipleChoiceInput `has the `IInputObject.GetShowProperty`() method, which returns a `MultipleChoiceProperty`, where you could change the `Choices`. 
But I see: the interface `IInputObject `is private. Maybe we should make it public? 
Or should we add a method to `MultipleChoiceInput `like `SetChoices`? Here we could access `multipleChoiceProperty.Choices`

closed 2026-10-10 by dsn27