FriendsOfCADability/CADability #256: 🚀 CADability 2.0 - Feature & Improvement Ideas

dsn27, 2025-03-19:
We are planning the next major version of CADability and would love to gather ideas for improvements, new features, and necessary refactorings. This issue serves as a central place for discussion and brainstorming.
✅ Current Ideas:

    📦 Create a NuGet Package
    Provide an easy way to install and manage CADability as a package.

    🔄 Remove dependency on Wintellect.PowerCollections
    Consider replacing it with .NET collections or another well-maintained alternative.

    📏 Better separation of GUI and Math Kernel
    Enhance modularity to make the core independent of the UI framework.

    🧪 Add more Unit Tests
    Increase test coverage to improve reliability and stability.

💡 Suggest Your Ideas!

If you have suggestions, please comment below with:

    A short description of the feature/improvement.
    Why it is needed.
    Any potential challenges or considerations.

Let’s make CADability 2.0 better together! 🚀

dsn27, 2025-03-20:
📌 Description

Adding support for DXF (including legacy versions) and DWG file formats would improve CADability’s interoperability with other CAD software. The implementation should follow a plugin-based approach, allowing future support for additional file formats with minimal changes to the core system.

💡 Why It’s Needed

-     DXF and DWG are widely used CAD formats, making CADability more versatile.
-     Supporting older DXF versions ensures compatibility with legacy systems.
-     A plugin-based approach future-proofs CADability by making it easier to extend with new formats without modifying the core codebase.

⚠️ Potential Challenges & Considerations

-     DXF and DWG have complex and evolving specifications, requiring careful implementation.
-     A modular plugin system must be well-designed to ensure seamless integration of additional file formats in the future.

SOFAgh, 2025-03-20:
- Description: 
Adapting the new DebuggerVisualizer Model.

- Why it's needed:
Debugging geometrical algorithms is practically impossible without the use of a visual debugger. I didn't update my VisualStudio because of that.

- Potential Challenges & Considerations
It is hard to debug the visual debugger.

stefan-tb, 2025-11-25:
Separation of the math kernel from the UI is highly desirable but massive task. The deep dependencies between the entire app shell classes (like Frame etc) and the GeoObjects makes it tough to support other platforms like the Web.