FriendsOfCADability/CADability #264: Border.GetParallel null exception

davidebazzi, 2025-04-23:
When an arc of the border will disappear for the computed offset there is an exception. 
Code to reproduce the bug:
`CADability.Shapes.BorderBuilder bb = new CADability.Shapes.BorderBuilder();`
`bb.AddSegment(new Line2D(new GeoPoint2D(2, 0), new GeoPoint2D(8, 0)));`
`bb.AddSegment(new Arc2D(new GeoPoint2D(8, 2), 2, Angle.A270, SweepAngle.Deg(90)));`
`bb.AddSegment(new Line2D(new GeoPoint2D(10, 2), new GeoPoint2D(10, 8)));`
`bb.AddSegment(new Arc2D(new GeoPoint2D(8, 8), 2, Angle.A0, SweepAngle.Deg(90)));`
`bb.AddSegment(new Line2D(new GeoPoint2D(8, 10), new GeoPoint2D(2, 10)));`
`bb.AddSegment(new Arc2D(new GeoPoint2D(2, 8), 2, Angle.A90, SweepAngle.Deg(90)));`
`bb.AddSegment(new Line2D(new GeoPoint2D(0, 8), new GeoPoint2D(0, 2)));`
`bb.AddSegment(new Arc2D(new GeoPoint2D(2, 2), 2, Angle.A180, SweepAngle.Deg(90)));`

`CADability.Shapes.Border b = bb.BuildBorder(false);`

`CADability.Shapes.Border[] b1 = b.GetParallel(-1.5, false, 0.001, 0); //It works.`
`CADability.Shapes.Border[] b2 = b.GetParallel(-2.5, false, 0.001, 0); //It generates and exception.`

davidebazzi, 2025-04-23:
Bug introduce with #257

davidebazzi, 2025-04-23:
Before Arc2D.Trim it would generate an arc of length 0, and it is handled with [line](https://github.com/FriendsOfCADability/CADability/blob/0fb6155412c58e17b7e728817153a5825adcbfdb/CADability/Border.cs#L2147) 
My suggestion is to add an `if(TheOnlyCurve != null)`

davidebazzi, 2025-04-23:
@SOFAgh What do you think there could be many other places to fix?

SOFAgh, 2025-04-23:
We would have to check all places, where ICurve2D.Trimm is used whether it is save, that Trimm could return null. Or we rely on the current behaviour, that Trimm returns curves with Length==0

dsn27, 2025-04-30:
I had a quick look.

It seems to be a mixed case. Some code checks for null, some for .Length > Precision.eps, and some not at all.

We can revert to the previous behavior where an invalid arc (.Length < Precision.eps) is returned.

Personally, I’m in favor of returning null for invalid arcs. It makes the invalid state explicit and avoids subtle bugs like the one I ran into with the smallest enclosing rectangle throwing a misleading error.

closed 2025-05-28 by dsn27

davidebazzi, 2026-01-22:
Importing a step file I encountered another place where it isn't check if the returned curve after a trim isn't null.
https://github.com/FriendsOfCADability/CADability/blob/959ebad71a28c41286a50cae10b16eeaa765c340/CADability/Border.cs#L2199

dsn27, 2026-01-22:
@davidebazzi Could you take a look and see if this Copilot suggestions fixes your bug and does not cause other issues for you?
You can get the new version here:
https://github.com/FriendsOfCADability/CADability/tree/copilot/fix-border-getparallel-null-exception

commit 11ee26a5 (2025-05-28): Beautify internal Border[] Split
Fix NRE if theOnlyCurve.Trim return null
Close #264